tscircuit/schematic-symbols #16: Simplify Structure for Hackoctoberest

seveibar, 2024-09-25:


imrishabh18, 2024-09-26:
One thing which can make contribution easier is, probably not having a big `symbols.svg`. Rather having individual named svg's, which the contributors can add in the assets folder according to the issue they have picked.

Another confusion which I faced was, that the .json files in the assets looks like we need to add a json file in there for the new component, which is not the case. Adding only the svg in `symbols.svg` will generate the json. So, probably moving those to a directory which says `generated` gives clear understanding (maybe adding it .gitignore will also help as this is generated everytime)

imrishabh18, 2024-09-27:
Okay, I will work upon these @seveibar

closed 2024-11-24 by imrishabh18