tscircuit/schematic-symbols #18: bun start not working, I think  wrong import format on windows

Abse2001, 2024-09-27:
[image]

seveibar, 2024-09-27:
CC @imrishabh18

seveibar, 2024-09-27:
i think this is an issue with [make-vfs](https://github.com/seveibar/make-vfs), it should use `import path from "path/posix"` instead of `import path from "path"` to be windows compatible. @Abse2001 would you like to take a stab at fixing it?

imrishabh18, 2024-09-27:
Taking a look

imrishabh18, 2024-09-27:
@Abse2001 Pull the latest main and let me know if it works

Abse2001, 2024-09-27:
Ok

Abse2001, 2024-09-27:
@imrishabh18  still the same problem
[image]

imrishabh18, 2024-09-27:
This seems weird, it should have not happened. Cause this file was not changed

imrishabh18, 2024-09-27:
Probably can you `git stash` your changes, and then checkout to the latest main and share what you see in this file?

closed 2024-09-27 by imrishabh18