tscircuit/schematic-symbols #79: Adding issue and PR templates

Sibam-Paul, 2024-10-01:
Will add bug ,docs, feat for issue templates
and PR templates

@tscircuitbot  **Please assign me this issue under hacktoberfest wanna work on it**

seveibar, 2024-10-01:
@Sibam-Paul for the sake of avoiding spam, we're only going to allow issue template PRs for people who are repeat contributors on a repo

Sibam-Paul, 2024-10-01:
@seveibar Why will it be a spam didn't get it ???

seveibar, 2024-10-01:
Only people who create a lot of good issues or solve a lot will create the best issue template for a repo. Generic issue templates aren't very helpful, a good issue template should be specific to the repo and make it easier for contributors and issue creators

closed 2024-10-01 by seveibar

Sibam-Paul, 2024-10-01:
@seveibar Got it bro thanks for explaining