tscircuit/schematic-symbols #171: Create symbol for Ground

imrishabh18, 2024-10-15:


TSP06, 2024-10-16:
@imrishabh18 I want to work on this please assign me this .

TSP06, 2024-10-16:
And also can you please label the issue Hacktoberfest

imrishabh18, 2024-10-16:
This repo has been labelled hacktoberfest. So, it will be counted towards it

TSP06, 2024-10-22:
[[image]
i have created the symbol but running bun run test 1 test is getting failed for ground_horz due to unmatched svg

TSP06, 2024-10-22:
@seveibar please help with this issue despite creating correct symbols bun run test is getting failed

seveibar, 2024-10-23:
@TSP06 you probably need to `export BUN_UPDATE_SNAPSHOTS=1 bun test <path/to/test/file.test.ts>`

You can also delete the original snapshot.

closed 2024-10-25 by imrishabh18