tscircuit/schematic-symbols #189: Diode component should have `up`, `down`, `left` and `right` variants

seveibar, 2024-11-10:
Look at diode symbol at https://symbols.tscircuit.com/

It should have a left/right/up/down variants instead of horz/vert

A utility function should be created and used to minimize effort when converting future symbols

/bounty $40

aybanda, 2024-11-10:
/attempt #189

aybanda, 2024-11-10:
Hi @seveibar,

I wanted to follow up on the diode variants discussed in this issue. I have implemented the following transformations based on the intended orientations:

I have created a utility function (i.e generateDiodeVarients.ts in utils directory) to generate diode_up.svg , diode_right, diode_left, diode_down in assets/symbols

Could you please confirm if these transformations align with your expectations for the diode orientations? I want to ensure that the implementation meets the project's requirements.

closed 2024-11-15 by anas-sarkez