tscircuit/schematic-symbols #205: Fix 90%+ of the Symbols Issues

seveibar, 2024-11-17:
https://github.com/user-attachments/assets/af8f3707-64fa-4afe-94b0-f3b2dfc0f28d

/bounty $60

hunxjunedo, 2024-11-20:
@seveibar the generating scripts are really buggy when it comes to keeping the right file names (messing between - and _ ). What should i do ? should I do the work keeping in mind that the generating scripts will be hardly used ? And wait for the scripts to be fixed later ? Because once you are done good and clean and you generate, you find yourself standing on zero, happened many times and its really exhausting.

hunxjunedo, 2024-11-20:
many symbols are not up-to-date with thier source svgs, what should I do in that scenario ? Like the constant-current-diode.

PR #211 (2024-11-23): fix symbols
fixes #205
/claim #205
related to #207

please confirm if all the symbols are okay so i can do the cleanup
closed 2024-11-23 by seveibar