tscircuit/schematic-symbols #224: Last recent updates text anchors are not acting as expected

Abse2001, 2024-12-15:
you can repro this in core update schematic-symbols and almost all tests will fail and give you these wrong text positions
[image]
[image]

seveibar, 2024-12-15:
/bounty $10

seveibar, 2024-12-15:
@Abse2001 can you incrementally install older versions of core until you find the version where this regression happened?

Abse2001, 2024-12-15:
> [@Abse2001](https://github.com/Abse2001) can you incrementally install older versions of core until you find the version where this regression happened?
@seveibar the issue started with schematic-symbols version 0.0.100 which breaks core tests 
and version 0.0.101 where the anchors began to act weird.
https://github.com/tscircuit/schematic-symbols/commit/2808d7a7526b6b4aa6137cf5678d483c058df092
https://github.com/tscircuit/schematic-symbols/commit/94a03f07a1c62cc9a2b7583f7b61300a326ca515